thrutheframe/Tiki3 #1: Red lizard

MichaelAronsen, 2021-09-09:
So I have a few red lizard hotends, but they have a screw in fitting for the ptfe tube that is too big, have you had your hands on a red lizard hotend?

thrutheframe, 2021-09-11:
I do not have a red lizard hotend, but I understand what you are talking about. The V6 fitting is too big. I do have a prototype replacement adapter designed to solve the issue. Would you like to print it and try to see if this solution fits? thanks

https://github.com/thrutheframe/Tiki3/tree/main/Red%20Lizard%20adapter

MichaelAronsen, 2021-09-11:
Thanks, will print it tonight.

I tinkercad'ed a version that sort of works, but I'm sure this is better.

MichaelAronsen, 2021-09-11:
So, it fits and lets the red lizard fits in the mount.

However, the bowden tube can slide out which isnt great.

I've attached the mangled stl's that I made with tinkercade to enlarge the
holes in the mount to fit the red lizard.

thrutheframe, 2021-09-12:
for the adapter to work,
1. the length of the ptfe tube must be slightly longer (0.5mm). When the orbiter is mounted, there will be a compression where the tube will be held in place.
2. The hole on the adapter need to be snug. I relooked my design and realise I gave a little too much allowance. I have change it from 4.4mm to 4.2mm (0.2mm is to compensate for any shrinkage after print) This shold help in the fitting.

I have also included the STEP and f3d files of the adapter if anyone wants to do any change to it.

MichaelAronsen, 2021-09-12:
Thanks!

I'm starting it out in a bowden setup with a voron m4 extruder on my ender
3, so need the bowden to be really secured..

thrutheframe, 2021-09-14:
ic. the STEP file of the body is available for editing to make the opening larger for the V6 attachment. I will close this issue as it is more or less solved. thanks

closed 2021-09-14 by thrutheframe