thrutheframe/Tiki3 #2: Cable Relief too long

Ramrawd, 2021-09-12:
Hi there, 

Really appreciate your work on this. 

On my setup with an ender 3 v2 the Cable Relief is too long for taller prints. It causes the cabling to get bunched up under the top bar and creates a lot of drag. 

I've designed a shorter version of this part in TinkerCad. It's pretty rough but it gets the job done. 

You may want to consider creating a shorter version of the cable guide for instances like mine.

Thanks! 
[3_2_Cable_relief_Short.zip](https://github.com/thrutheframe/Tiki3/files/7149001/3_2_Cable_relief_Short.zip)
[image]

thrutheframe, 2021-09-12:
Thank you for the feedback. I have designed a short version (20mm shorter) of the cable relief. U can download it from the folder.
https://github.com/thrutheframe/Tiki3/tree/main/3_2%20other%20parts

Ramrawd, 2021-09-12:
Perfect! Thanks so much for doing that!

closed 2021-09-12 by Ramrawd