thrutheframe/Tiki3 #7: Orbiter Extruder V2.0

Ramrawd, 2021-12-16:
Hi there, 

Any chance you could add a new mounting bracket for the orbiter v2.0 extruder? 

https://drive.google.com/drive/folders/1Kr8K3wX7nMKRvu9yo7Z1blTNQCAX_nJb

There are files for the new design. 

Pre-orders are happening here: https://lukeslabonline.com/products/orbiter-v2-dual-drive-direct-extruder

Thanks!

thrutheframe, 2021-12-16:
As of now, I am not planning to design any mount for the Orbiter V2.0. Reasons:

1. The distance and position of mounting holes have changed with V2.0. Tiki3 will have to be redesigned from the ground up, as it uses the asymmetric mounting position of V1.5 to hold the bracket together.  
2. My existing Orbiter V1.5 is working fine and I don't see myself buying the V2.0 as an upgrade. 
3. I want to spend time and resources working on another design project.

If you are looking for a mounting system for V2.0, I think Robert will be uploading something soon and the maker of HeroMe had done one too.

Ramrawd, 2021-12-16:
Ah alright! Not a problem. Thanks for the explanation. 

As always, really appreciate you work. I'll be keeping an eye out for other adaptations. 

Have a good one!

thrutheframe, 2021-12-16:
I will be looking for testers for my other designer project if you are interested. u can message me on facebook.

Ramrawd, 2021-12-16:
Ah alright! I don't have a facebook account but I'll troll your githubs and see if anything new pops up. You're always welcome to message me if you'd like as well!

thrutheframe, 2021-12-16:
ic. sure, will msg u when it time to look for testers...lol thanks again

closed 2022-01-14 by thrutheframe